ASCOMInitiative/ConformU #3: Confusing issue message in discovery diagnostics

RReverser, 2023-02-23:
While running discovery diagnostics against my sample server, I'm getting a report with 2 issues. The log looks like following:

```
Found response from 192.168.133.22                 => {"AlpacaPort":3000}
                                                      OK    - The Alpaca port JSON key name is spelled and cased correctly
                                                      OK    - Successfully parsed the JSON response and extracted port number: 3000.
                                                      OK    - Successfully read API version number: 1. Testing the Management API...
                                                      OK    - Successfully read device description
                                                      ISSUE - Exception reading device description: Object reference not set to an instance of an object.
                                                      OK    - Successfully read configured devices response
                                                                Found Camera device.
                                                                  Device number: 0
                                                                  Device name: My Camera
                                                                  Unique ID: ffe84935-e951-45b3-9835-d532b04ee932
 
Found response from 127.0.0.1                      => {"AlpacaPort":3000}
                                                      OK    - The Alpaca port JSON key name is spelled and cased correctly
                                                      OK    - Successfully parsed the JSON response and extracted port number: 3000.
                                                      OK    - Successfully read API version number: 1. Testing the Management API...
                                                      OK    - Successfully read device description
                                                      ISSUE - Exception reading device description: Object reference not set to an instance of an object.
                                                      OK    - Successfully read configured devices response
                                                                Found Camera device.
                                                                  Device number: 0
                                                                  Device name: My Camera
                                                                  Unique ID: ffe84935-e951-45b3-9835-d532b04ee932
 
Issue Summary
192.168.133.22:3000                                    Exception reading device description: Object reference not set to an instance of an object.
127.0.0.1:3000                                         Exception reading device description: Object reference not set to an instance of an object.
```

The message "Exception reading device description: Object reference not set to an instance of an object." seems confusing as it doesn't provide any details about what went wrong, especially given that the calls before and after succeeded.

I'll probably get to the bottom of it but would be nice to improve diagnostics message here.

RReverser, 2023-02-23:
FWIW I found what it was in my case - a silly mistake where `/management/v1/description` was returning `Location`, `Manufacturer`, etc. as top-level fields instead of contained inside a nested `{"Value": {...}}` object.

Hopefully this helps in improving the error message.

Peter-Simpson, 2023-02-26:
Hi Ingvar,

Thanks, your report is helpful, I'm glad you found the issue.

I've added traps for this and similar issues to ConformU, which will now provide more helpful messages that include the returned JSON.

Best wishes, Peter

commit 40ddc50b (2023-02-26): Improved discovery diagnostics messaging when the device does not return expected JSON responses. Fixes #3.
closed 2023-02-26 by Peter-Simpson