ASCOMInitiative/ConformU #4: PUT requests are checked for case sensitivity while spec says all params are case-insensitive

RReverser, 2023-02-23:
When running against my server, in Alpaca protocol checks I noticed issues like the following:

```
PUT AbortExposure ==> Different ClientId case - Expected HTTP status : 400 (BadRequest) but received status: 200 (OK).
  Response: {"ClientTransactionID":67890,"ServerTransactionID":270,"ErrorMessage":"","ErrorNumber":0}

PUT AbortExposure ==> Different ClientTransactionId case - Expected HTTP status : 400 (BadRequest) but received status: 200 (OK).
  Response: {"ClientTransactionID":67890,"ServerTransactionID":271,"ErrorMessage":"","ErrorNumber":0}
```

However, the [ASCOM Alpaca API Reference](https://ascom-standards.org/Developer/ASCOM%20Alpaca%20API%20Reference.pdf) says the following:

> ## 2.2 Case Sensitivity
> ### 2.2.2 Alpaca API Parameters
> Alpaca parameters are key-value pairs where:
>  - The parameter key is case insensitive
>  - The parameter value can have any casing required.
>
> [...]
>
> Clients and drivers must expect incoming API parameter keys to have arbitrary casing.

It doesn't mention any differentiation between GET and PUT requests and, indeed, given that the format of the parameters is the same (`Param=1&AnotherParam=2`), only served in the query string for GET requests vs in the body for PUT requests, it would be odd for the case sensitivity requirements to differ as many servers use the same parser for this format.

RReverser, 2023-02-28:
@Peter-Simpson Can you please advise on whether it's the spec text or the ConformU implementation that is right in this scenario?

RReverser, 2023-03-13:
Ping.

Peter-Simpson, 2023-03-14:
Hi Ingvar,

Sorry for the delay in replying. 

It looks like you have an early version of the API reference document, the casing requirements were clarified in the 2021 version 4 of the API spec and we are now on version 7: [Alpaca API Reference v7](https://github.com/ASCOMInitiative/ASCOMRemote/blob/master/Documentation/ASCOM%20Alpaca%20API%20Reference.pdf). Can you let me know where you found the version you are using so that we can update it to the current version?

I made query parameters case insensitive so that developers can use all lower case for convenience. However, this introduced an inconsistency with form parameter handling that I didn't spot at the time, and so we are where we are with case insensitive query parameters and case sensitive form parameters.

Best wishes, Peter

RReverser, 2023-03-18:
@Peter-Simpson, one more follow-up question:

> Alpaca parameters are key-value pairs where:
• The parameter key is case insensitive
• The parameter value can have any casing required.

What is the difference between two statements? In particular, not sure what it means for parameter value to be allowed to have any casing.

Does that mean that strings, e.g. `SensorName` in `sensordescription` command, need to be matched case-insensitively? Or does it apply only to types like `bool` and enums?

Peter-Simpson, 2023-03-18:
I assume this text is from section 2.2.2 on page 6 of API Reference v7.  What this means is that senders can use any casing they wish for parameter names and that recipients must process parameter names into a known casing state such all lower case or all upper case before attempting a comparison with expected parameter names. 

The parameter value statement is a recognition that any non-numeric parameter values can also use any casing.

The second bullet may actually be redundant for Alpaca because, off the top of my head, I can't think of any string parameter values that are passed as query parameters.

RReverser, 2023-03-19:
> I assume this text is from section 2.2.2 on page 6 of API Reference v7.

Yes, sorry, that's the one.

> The parameter value statement is a recognition that any non-numeric parameter values can also use any casing.

So just to clarify, does that mean that paragraph is synonymous to "Both parameter keys and values are case-insensitive." or is there some subtle difference between handling of the two?

> The second bullet may actually be redundant for Alpaca because, off the top of my head, I can't think of any string parameter values that are passed as query parameters.

Please see above - I mentioned "`SensorName` in `sensordescription` command" as an example, and the only other example is also `SensorName` in `timesincelastupdate` command.

If parameter values are meant to use any casing / be case-insensitive, the implication seems to be that SensorName needs to be matched in case-insensitive way against all known sensors.

If that's not the intent, perhaps better to rephrase that section to say something like "booleans can be sent in any casing in query parameters", since that's the only other data type that is sent as a string.

Peter-Simpson, 2023-03-19:
The subtle difference between the two statements is that case sensitivity for the key name is defined in the Alpaca API Reference specification (as case insensitive) whereas, for parameters that are part of an ASCOM interface, any case sensitivity is defined in the device's ASCOM Interface specification.

A more comprehensive version of the second bullet would be:
- Integer parameter values should use Arabic numerals without thousands separators (.NET invariant culture).
- Floating point parameter values should use Arabic numerals with point '.' as the decimal separator and without thousands separators  (.NET invariant culture).
- Boolean parameter values should be sent in text form: **_true_** or **_false_**
- String parameter values that are defined in an ASCOM Interface, such as the SensorName parameter in  IObservingConditions.SensorDescription(string SensorName), must be cased in accordance with any requirements stated in the Interface specification.

I had a look at the SensorDescription and and TimeSinceLastUpdate defininitions and netiher of them specify any kind of case behaviour so we should probably clarify that. For now, any casing should be accepted.

Thanks for calling out the lack of clarity.

Best wishes, Peter

RReverser, 2023-03-19:
> whereas, for parameters that are part of an ASCOM interface, any case sensitivity is defined in the device's ASCOM Interface specification.

Ah, thanks, this clicked for me.

> A more comprehensive version of the second bullet would be:

Right, I'm not as concerned about numeric parameters (although I guess that is helpful to someone too), as about string-like - namely, String itself just for SensorName and boolean.

You mentioned here that booleans are sent as `true` or `false` which also matches what JavaScript and Rust do for the default string representation; but I noticed that ConformU sends them as `True` or `False`.

So

Peter-Simpson, 2023-03-21:
Yes that was one I missed. At this point I don't think we can tighten the specification, we just have to use a broad definition. Here is an updated version:

- Integer parameter values should use Arabic numerals without thousands separators (.NET invariant culture).
- Floating point parameter values should use Arabic numerals with point '.' as the decimal separator and without thousands separators (.NET invariant culture).
- Bolean parameter values should be sent in text form: **_true_** or **_false_** and can use any casing.
- String parameter values that are defined in an ASCOM Interface, such as the SensorName parameter in IObservingConditions.SensorDescription(string SensorName), must be cased in accordance with any requirements stated in the Interface specification.

closed 2023-03-21 by RReverser