ASCOMInitiative/ConformU #5: Feature request: JSON logs

RReverser, 2023-02-24:
Right now, ConformU can work in command line and produce a JSON of the resulting issues. However, those can be hard to understand without the surrounding context.

Instead, I found that running a server and ConformU in the same console in parallel is more useful as it allows to sea interleaved logs, where ConformU outputs OK/ISSUE for each test, and right above each such test I can see the more detailed from my server and understand what caused corresponding issue.

I'd like to take that a step further and integrate those ConformU logs into the test runner so that it could report each OK/ISSUE as if they were regular tests. I think this would benefit not only Rust, but also test runners in other languages (e.g. Python for Alpyca). For that, ideally I'd avoid parsing the logs with a regex and instead ask for ConformU to output them as individual JSON objects when e.g. `--commandline --json-log` is used.

RReverser, 2023-02-24:
I thought this might be something I could implement myself in a PR, but I'm having trouble building ConformU locally. Looks like it depends on some unreleased versions of ASCOM packages that are not on nuget yet:

```
c:\Users\me\Documents\ConformU\ConformU\ConformU.csproj(0,0): Error: Unable to find package ASCOM.Exception.Library with version (>= 6.6.2-rc.1)
  - Found 1 version(s) in nuget.org [ Nearest version: 6.6.1 ]
  - Found 0 version(s) in Microsoft Visual Studio Offline Packages
c:\Users\me\Documents\ConformU\ConformU\ConformU.csproj(0,0): Error: Unable to find package ASCOM.Com.Components with version (>= 1.0.94)
  - Found 16 version(s) in nuget.org [ Nearest version: 1.0.70 ]
  - Found 0 version(s) in Microsoft Visual Studio Offline Packages
c:\Users\me\Documents\ConformU\ConformU\ConformU.csproj(0,0): Error: Unable to find package ASCOM.Alpaca.Components with version (>= 1.0.94)
  - Found 17 version(s) in nuget.org [ Nearest version: 1.0.70 ]
  - Found 0 version(s) in Microsoft Visual Studio Offline Packages
c:\Users\me\Documents\ConformU\ConformU\ConformU.csproj(0,0): Error: Unable to find package ASCOM.Tools with version (>= 1.0.94)
  - Found 16 version(s) in nuget.org [ Nearest version: 1.0.70 ]
  - Found 0 version(s) in Microsoft Visual Studio Offline Packages
```

Peter-Simpson, 2023-02-26:
Hi Ingvar,

 

I’ve uploaded the latest versions of the ASCOM Library and ASCOM Exceptions to NuGet and updated ConformU to use them, hopefully you will be able to build the application now. The Library package is release and the exceptions package is pre-release.

 

Best wishes, Peter

RReverser, 2023-02-28:
Thanks, I can confirm I can build now. Will try to carve out time and add JSON logs here.

RReverser, 2024-08-19:
I'm going to close this for now since it's clear I won't have time to implement it anytime soon...

closed 2024-08-19 by RReverser