ASCOMInitiative/ConformU #8: LastExposureDuration is not the same as image duration

RReverser, 2023-04-21:
When trying to report an actual exposure duration, I'm getting ConformU test failure:

```
03:40:43.398 LastExposureDuration                ISSUE    LastExposureDuration is not the same as image duration: 2.0668501 2
```

I could simply set it to whatever user provided in `duration` field, but the [ASCOM docs](https://ascom-standards.org/Help/Developer/html/P_ASCOM_DeviceInterface_ICameraV3_LastExposureDuration.htm) for this property say:

> This may differ from the exposure time requested due to shutter latency, camera timing precision, etc.

Given that this is meant to be an actual exposure time including latency, I think ConformU should have some error tolerance in this test, similarly to how it already has error tolerance for `LastExposureStartTime` test.

Peter-Simpson, 2023-04-21:
Hi Ingvar,

 

The LastExposureDuration test in ConformU has a 2% tolerance, so values in the range 1.96 to 2.04 seconds are accepted for a 2 second exposure. To date, I’ve not had reports from other people that this tolerance is too tight.

 

If the time difference from the expected value is essentially a constant offset, you should be able to get your device to pass by configuring a longer test exposure time on ConformU’s camera settings page.

 

Hope this helps,

 

Best wishes, Peter

RReverser, 2023-04-21:
Ah, thanks, sounds good. The messaging gave me an impression that it was doing an exact comparison.

I think longer exposure in settings should cover it for me.

closed 2023-04-21 by RReverser