ASCOMInitiative/ConformU #9: conform output question

ivonnyssen, 2023-12-22:
Hi,
I am working on an alpaca driver for qhyccd cameras. Thank you for ConformU - it has been a tremendous help!!

I am not sure about the output below though - it looks like the driver is correctly rejecting the invalid values, but they seem to be flagged as issues? Am I sending the wrong response?

Is there a way in which I could figure out what my driver should be sending?

Thank you so much!
Igor

15:55:42.703 ASCOM Universal Device Conformance Checker Version 2.0.0.16486, Build time: Sun 28 May 2023 23:03:20
15:55:42.704 
15:55:42.704 Operating system is Linux 5.15.0-91-generic #101-Ubuntu SMP Tue Nov 14 13:30:08 UTC 2023 64bit, Application is 64bit.
15:55:42.704 
15:55:42.704 Alpaca device: QHYCCD-QHY5III290C-1caff64da5f739b72 (127.0.0.1:8000 Camera/0)
15:55:42.704 
15:55:42.704 CreateDevice                        INFO     Creating Alpaca device: IP address: 127.0.0.1, IP Port: 8000, Alpaca device number: 0
15:55:42.704 CreateDevice                        INFO     Alpaca device created OK
15:55:43.705 CreateDevice                        INFO     Successfully created driver
15:55:43.705 ConformanceCheck                    OK       Driver instance created successfully
15:55:43.705 
15:55:43.705 Connect to device
15:55:44.868 Connected                           OK       True
15:55:44.868 
15:55:44.868 Common Driver Methods
15:55:44.870 InterfaceVersion                    OK       3
15:55:44.870 Connected                           OK       True
15:55:44.872 Description                         OK       QHYCCD camera
15:55:44.873 DriverInfo                          OK       qhyccd_alpaca driver
15:55:44.874 DriverVersion                       OK       0.1.0
15:55:44.875 Name                                OK       QHYCCD-QHY5III290C-1caff64da5f739b72
15:55:44.875 Action                              INFO     Conform cannot test the Action method
15:55:44.876 SupportedActions                    OK       Driver returned an empty action list
15:55:44.877 
15:55:44.877 Can Properties
15:55:44.878 CanAbortExposure                    OK       True
15:55:44.879 CanAsymmetricBin                    OK       False
15:55:44.880 CanGetCoolerPower                   OK       False
15:55:44.881 CanPulseGuide                       OK       False
15:55:44.882 CanSetCCDTemperature                OK       False
15:55:44.883 CanStopExposure                     OK       False
15:55:44.884 CanFastReadout                      OK       False
15:55:44.884 
15:55:44.884 Pre-run Checks
15:55:44.884 
15:55:44.884 Last Tests
15:55:44.886 LastExposureDuration                OK       LastExposureDuration returned an error before an exposure was made: A value has not been set get - no value has been set.
15:55:44.887 LastExposureStartTime               OK       LastExposureStartTime returned an error before an exposure was made: A value has not been set get - no value has been set.
15:55:44.887 
15:55:44.887 Properties
15:55:44.895 MaxBinX                             OK       2
15:55:44.896 MaxBinY                             OK       2
15:55:44.896 BinX Read                           OK       1
15:55:44.897 BinY Read                           OK       1
15:55:44.898 BinX Write                          OK       Received error on setting BinX to 0: bin value must be >= 1
15:55:44.901 BinX Write                          OK       Received error on setting BinX to 3: A value has not been set get - no value has been set.
15:55:44.902 BinY Write                          OK       Received error on setting BinY to 0: bin value must be >= 1
15:55:44.904 BinY Write                          OK       Received error on setting BinY to 3: A value has not been set get - no value has been set.
15:55:44.914 BinXY Write                         OK       Successfully set symmetric XY binning: 1 x 1
15:55:44.916 BinXY Write                         OK       Successfully set symmetric XY binning: 2 x 2
15:55:44.919 CameraState                         OK       Idle
15:55:44.919 CameraXSize                         OK       1920
15:55:44.920 CameraYSize                         OK       1080
15:55:44.921 CCDTemperature                      OK       Optional member returned a PropertyNotImplementedException error.
15:55:44.923 CoolerOn Read                       OK       Optional member returned a PropertyNotImplementedException error.
15:55:44.935 CoolerOn Read                       OK       Optional member returned a PropertyNotImplementedException error.
15:55:44.936 CoolerPower                         OK       Optional member returned a PropertyNotImplementedException error.
15:55:44.939 ElectronsPerADU                     OK       Optional member returned a PropertyNotImplementedException error.
15:55:44.940 FullWellCapacity                    OK       Optional member returned a PropertyNotImplementedException error.
15:55:44.941 HasShutter                          OK       False
[... 86 lines cut ...]
15:55:50.195 Reject Bad YSize (bin 2 x 2)        ISSUE    Unable to set NumY to 541: Invalid value
15:55:50.196 Reject Bad YSize (bin 2 x 2)        INFO     Exposure test abandoned because the camera is not idle or at least one of BinX, BinY, NumX, NumY, StartX or StartY could not be set.
15:55:50.199 Reject Bad XStart (bin 2 x 2)       ISSUE    Unable to set StartX to 961: A value has not been set get - no value has been set.
15:55:50.203 Reject Bad XStart (bin 2 x 2)       INFO     Exposure test abandoned because the camera is not idle or at least one of BinX, BinY, NumX, NumY, StartX or StartY could not be set.
15:55:50.219 Reject Bad YStart (bin 2 x 2)       ISSUE    Unable to set StartY to 541: A value has not been set get - no value has been set.
15:55:50.220 Reject Bad YStart (bin 2 x 2)       INFO     Exposure test abandoned because the camera is not idle or at least one of BinX, BinY, NumX, NumY, StartX or StartY could not be set.
15:55:50.221 
15:55:50.221 Post-run Checks
15:55:50.240 
15:55:50.241 Disconnect from device
15:55:50.867 Connected                           OK       False
15:55:50.867 
15:55:50.867 Conformance test has finished
15:55:50.867 
15:55:50.868 Your device had 8 issues, 0 errors and 0 configuration alerts
15:55:50.868 
15:55:50.868 Issue Summary
15:55:50.868 Reject Bad XSize (bin 1 x 1)        ISSUE    Unable to set NumX to 1921: Invalid value
15:55:50.868 Reject Bad YSize (bin 1 x 1)        ISSUE    Unable to set NumY to 1081: Invalid value
15:55:50.868 Reject Bad XStart (bin 1 x 1)       ISSUE    Unable to set StartX to 1921: Invalid value
15:55:50.868 Reject Bad YStart (bin 1 x 1)       ISSUE    Unable to set StartY to 1081: A value has not been set get - no value has been set.
15:55:50.868 Reject Bad XSize (bin 2 x 2)        ISSUE    Unable to set NumX to 961: Invalid value
15:55:50.868 Reject Bad YSize (bin 2 x 2)        ISSUE    Unable to set NumY to 541: Invalid value
15:55:50.868 Reject Bad XStart (bin 2 x 2)       ISSUE    Unable to set StartX to 961: A value has not been set get - no value has been set.
15:55:50.868 Reject Bad YStart (bin 2 x 2)       ISSUE    Unable to set StartY to 541: A value has not been set get - no value has been set.

ivonnyssen, 2023-12-27:
found the answer to the question here: https://ascom-standards.org/alpyca/alpaca.camera.html

The values for StartX, StartY, NumX and NumY should be checked for validity when starting an exposure, not when they are being set.

closed 2023-12-27 by ivonnyssen

Peter-Simpson, 2023-12-28:
Glad you found the answer! It is this way because clients can easily set an "illegal" configurations while they are setting up the next exposure, before all the exposure parameters have been set to their new "correct" values.

For example, say the previous exposure was at bin 3 while the next exposure is at bin 1. Also suppose that the client changes NumX first, to be the full width of a Bin x1 image. If validation is done immediately, based on the current settings of other parameters that are still at their bin x3 settings, this new value would be "invalid".

Hence the need to check validation only after all parameters are set i.e. when the exposure is started.

Best wishes Peter