ASCOMInitiative/ConformU #10: sampling of declinationrate is too short

charleslemaire0, 2023-12-29:
Hi,

During the "extended rate offset tests" the duration to evalute if the rate 0.1 or 1.0 for declination is precise (with 5% tolerance) is too short. Could you increase the time or make it avaible on the UI. It is also interesting if some test happens at the same side of pier.
Strange is that for the first goto the declination rate is set to 0.1 but for the other goto the declination rate is 0. Maybe this is a small mistake?

Thanks

Charles for the TeenAstro Group

Peter-Simpson, 2023-12-31:
Hi Charles,

I can't remember whether I came to the same conclusion about test duration or whether someone else mentioned it, but I have already added an ability to configure the test time in the Telescope settings tab. Here is a link to the latest test version that contains the change: [Conform Universal 2.0.1 rc 7](https://download.ascom-standards.org/beta/ConformU(2.0.1-rc.7+23374)Setup.exe).

I don't fully understand  what you are asking for when you say "It is also interesting if some test happens at the same side of pier" and your other point about "Strange is that for the first goto the declination rate is set to 0.1 but for the other goto the declination rate is 0".

Please can you expand on these, first saying what you actually see happen and then how this could be done differently, or what else could be done too.

Many thanks, Peter

charleslemaire0, 2024-01-01:
Hi Peter,

The rc works for me as expected for these topics! Thanks

closed 2024-01-01 by charleslemaire0