ASCOMInitiative/ConformU #11: RightAscensionRate test

charleslemaire0, 2024-01-01:
Hello  Peter

Here a small issue:

When CanSetRightAscensionRate  is true and CanSetDeclinationRate is false, the rate test routines tries anyway to set DeclinationRate in the write RightAscensionRate test routine, and report an false positive issue.

attached you can find the log

Thanks for your time and great work, this tool has so much value!

12:45:57.347 ASCOM Universal Device Conformance Checker Version 2.0.1.23374, Build time: Sat 30 December 2023 14:28:12
12:45:57.384 
12:45:57.400 Operating system is Microsoft Windows 10.0.19045 64bit, Application is 64bit.
12:45:57.441 
12:45:57.444 COM Driver ProgID: ASCOM.TeenAstro.Telescope
12:45:57.448 
12:45:57.469 CreateDevice                        INFO     Creating NATIVE COM device: ASCOM.TeenAstro.Telescope
12:45:57.651 CreateDevice                        INFO     Successfully created driver
12:45:58.674 CreateDevice                        OK       Driver instance created successfully
12:45:58.675 
12:45:58.676 Connect to device
12:45:58.680 InterfaceVersion                             About to get property InterfaceVersion
12:45:58.694 Connected                                    About to set Connected property
12:45:58.761 Connect                                      About to get Connected property
12:45:58.772 Connected                           OK       Connected to device successfully using Connected = True
12:45:58.773 
12:45:58.781 Common Driver Methods
12:45:58.781 InterfaceVersion                    OK       3
12:45:58.784 Connected                                    About to get property Connected
12:45:58.789 Connected                           OK       True
12:45:58.791 Description                                  About to get property Description
12:45:58.803 Description                         OK       TeenAstro Telescope
12:45:58.804 DriverInfo                                   About to get property DriverInfo
12:45:58.809 DriverInfo                          OK       Version: 1.5
12:45:58.809 DriverVersion                                About to get property DriverVersion
12:45:58.814 DriverVersion                       OK       1.5
12:45:58.815 Name                                         About to get property Name
12:45:58.821 Name                                OK       TeenAstro
12:45:58.821 
12:45:58.821 Action                              INFO     Conform cannot test the Action method
12:45:58.822 SupportedActions                             About to call method SupportedActions
12:45:58.847 SupportedActions                    OK       Found action: AutoAlign
12:45:58.868 
12:45:58.868 DeviceState                         INFO     DeviceState tests omitted - DeviceState is not available in this interface version.
12:45:58.871 
12:45:58.872 Can Properties
12:45:58.880 CanFindHome                                  About to get CanFindHome property
12:45:58.891 CanFindHome                         OK       True
12:45:58.900 CanPark                                      About to get CanPark property
12:45:58.914 CanPark                             OK       True
12:45:58.917 CanPulseGuide                                About to get CanPulseGuide property
12:45:58.929 CanPulseGuide                       OK       True
12:45:58.931 CanSetDeclinationRate                        About to get CanSetDeclinationRate property
12:45:58.947 CanSetDeclinationRate               OK       False
12:45:58.948 CanSetGuideRates                             About to get CanSetGuideRates property
12:45:58.954 CanSetGuideRates                    OK       True
12:45:58.956 CanSetPark                                   About to get CanSetPark property
12:45:58.962 CanSetPark                          OK       True
12:45:58.974 CanSetPierSide                               About to get CanSetPierSide property
12:45:58.993 CanSetPierSide                      OK       True
12:45:59.005 CanSetRightAscensionRate                     About to get CanSetRightAscensionRate property
12:45:59.023 CanSetRightAscensionRate            OK       True
12:45:59.024 CanSetTracking                               About to get CanSetTracking property
12:45:59.028 CanSetTracking                      OK       True
12:45:59.028 CanSlew                                      About to get CanSlew property
12:45:59.034 CanSlew                             OK       True
12:45:59.034 CanSlewltAz                                  About to get CanSlewAltAz property
12:45:59.041 CanSlewltAz                         OK       True
12:45:59.042 CanSlewAltAzAsync                            About to get CanSlewAltAzAsync property
12:45:59.049 CanSlewAltAzAsync                   OK       True
[... 66 lines cut ...]
12:46:00.531 RightAscensionRate Write                     Set rate to 0.0 - About to get Slewing property
12:46:00.546 RightAscensionRate Write                     Set rate to 0.0 - About to set RightAscensionRate property to 0
12:46:01.571 RightAscensionRate Write                     Set rate to 0.0 - About to get RightAscensionRate property
12:46:01.582 RightAscensionRate Write                     Set rate to 0.0 - About to get the Slewing property
12:46:01.591 RightAscensionRate Write            OK       Set rate to 0.0 - successfully set rate to 0
12:46:01.592 RightAscensionRate Write                     Set rate to 0.1 - About to get Slewing property
12:46:01.604 RightAscensionRate Write                     Set rate to 0.1 - About to get Slewing property
12:46:01.616 RightAscensionRate Write                     Set rate to 0.1 - About to set RightAscensionRate property to 0.1
12:46:02.662 RightAscensionRate Write                     Set rate to 0.1 - About to get RightAscensionRate property
12:46:02.671 RightAscensionRate Write                     Set rate to 0.1 - About to get the Slewing property
12:46:02.690 RightAscensionRate Write            OK       Set rate to 0.1 - successfully set rate to 0.1
12:46:02.720 RightAscensionRate Write -9.0                About to get SiteLatitude property
12:46:02.731 RightAscensionRate Write -9.0                About to get SiteLongitude property
12:46:02.744 RightAscensionRate Write -9.0                About to get SiteElevation property
12:46:02.814 SlewScope                                    About to set Tracking property to true
12:46:02.863 SlewScope                                    About to call SlewToCoordinatesAsync method, RA: +03:18:39.33, Declination: +85:00:00.0
12:46:02.946 Hour angle -9.00                             About to get Slewing property multiple times
12:46:07.983 SlewScope                                    About to get SideOfPier
12:46:08.000 SlewScope                                    About to get SideOfPier property
12:46:08.016 RightAscensionRate Write -9.0                About to get SideOfPier property
12:46:08.046 RightAscensionRate Write -9.0                About to get RightAscension property
12:46:08.054 RightAscensionRate Write -9.0                About to get Declination property
12:46:08.062 RightAscensionRate Write -9.0                About to set RightAscensionRate property
12:46:08.069 RightAscensionRate Write -9.0                About to set DeclinationRate property
12:46:08.092 RightAscensionRate Write            ISSUE    Unexpected error - CanSetRightAscensionRate is True: Set DeclinationRate via :SXRd,0 has failed

Peter-Simpson, 2024-01-02:
Hi Charles,

Thanks for reporting this issue. I've revised the tests to omit testing an axis rate offset if the associated CanXXX method is false.

Please can you let me know whether this new version fixes the issue. [ConformU v 2.0.1 RC 8](https://download.ascom-standards.org/beta/ConformU(2.0.1-rc.8+23508)Setup.exe) 

Best wishes, Peter

charleslemaire0, 2024-01-09:
It works!

closed 2024-01-09 by charleslemaire0