ASCOMInitiative/ConformU #13: Abortslew check when Canslew is false

charleslemaire0, 2024-02-17:
Hello,
TeenAstro can be configure as pure PUSH-TO telescope. I set in my driver Canslew to false when the telescope is configured as PUSH-TO
but I get this message in the U conform:

15:06:30.390 ASCOM Universal Device Conformance Checker Version 2.1.0.23787, Build time: Thu 11 January 2024 11:25:50
15:06:30.413 
15:06:30.414 Operating system is Microsoft Windows 10.0.19045 64bit, Application is 64bit.
15:06:30.414 
15:06:30.415 COM Driver ProgID: ASCOM.TeenAstro.Telescope
15:06:30.422 
15:06:30.428 CreateDevice                        INFO     Creating NATIVE COM device: ASCOM.TeenAstro.Telescope
15:06:30.776 CreateDevice                        INFO     Successfully created driver
15:06:31.803 CreateDevice                        OK       Driver instance created successfully
15:06:31.806 
15:06:31.809 Connect to device
15:06:31.814 InterfaceVersion                             About to get property InterfaceVersion
15:06:31.898 Connected                                    About to set Connected property
15:06:31.955 Connect                                      About to get Connected property
15:06:31.964 Connected                           OK       Connected to device successfully using Connected = True
15:06:31.969 
15:06:31.980 Common Driver Methods
15:06:31.981 InterfaceVersion                    OK       3
15:06:31.989 Connected                                    About to get property Connected
15:06:31.993 Connected                           OK       True
15:06:31.995 Description                                  About to get property Description
15:06:32.008 Description                         OK       TeenAstro Telescope
15:06:32.010 DriverInfo                                   About to get property DriverInfo
15:06:32.022 DriverInfo                          OK       Version: 1.5
15:06:32.025 DriverVersion                                About to get property DriverVersion
15:06:32.036 DriverVersion                       OK       1.5
15:06:32.041 Name                                         About to get property Name
15:06:32.046 Name                                OK       TeenAstro
15:06:32.047 
15:06:32.048 Action                              INFO     Conform cannot test the Action method
15:06:32.048 SupportedActions                             About to call method SupportedActions
15:06:32.076 SupportedActions                    OK       Found action: AutoAlign
15:06:32.079 
15:06:32.088 DeviceState                         INFO     DeviceState tests omitted - DeviceState is not available in this interface version.
15:06:32.092 
15:06:32.095 Can Properties
15:06:32.107 CanFindHome                                  About to get CanFindHome property
15:06:32.114 CanFindHome                         OK       False
15:06:32.123 CanPark                                      About to get CanPark property
15:06:32.160 CanPark                             OK       False
15:06:32.161 CanPulseGuide                                About to get CanPulseGuide property
15:06:32.171 CanPulseGuide                       OK       False
15:06:32.174 CanSetDeclinationRate                        About to get CanSetDeclinationRate property
15:06:32.187 CanSetDeclinationRate               OK       False
15:06:32.192 CanSetGuideRates                             About to get CanSetGuideRates property
15:06:32.203 CanSetGuideRates                    OK       False
15:06:32.207 CanSetPark                                   About to get CanSetPark property
15:06:32.221 CanSetPark                          OK       False
15:06:32.226 CanSetPierSide                               About to get CanSetPierSide property
15:06:32.234 CanSetPierSide                      OK       True
15:06:32.243 CanSetRightAscensionRate                     About to get CanSetRightAscensionRate property
15:06:32.255 CanSetRightAscensionRate            OK       False
15:06:32.256 CanSetTracking                               About to get CanSetTracking property
15:06:32.278 CanSetTracking                      OK       False
15:06:32.279 CanSlew                                      About to get CanSlew property
15:06:32.298 CanSlew                             OK       False
15:06:32.307 CanSlewltAz                                  About to get CanSlewAltAz property
15:06:32.320 CanSlewltAz                         OK       False
15:06:32.324 CanSlewAltAzAsync                            About to get CanSlewAltAzAsync property
15:06:32.342 CanSlewAltAzAsync                   OK       False
15:06:32.345 CanSlewAsync                                 About to get CanSlewAsync property
15:06:32.359 CanSlewAsync                        OK       False
15:06:32.360 CanSync                                      About to get CanSync property
15:06:32.374 CanSync                             OK       True
15:06:32.375 CanSyncAltAz                                 About to get CanSyncAltAz property
15:06:32.381 CanSyncAltAz                        OK       True
[... 145 lines cut ...]
System.Runtime.InteropServices.COMException (0x8004040C): Unknown  Action is not implemented in this driver.
   at Microsoft.CSharp.RuntimeBinder.ComInterop.ComRuntimeHelpers.CheckThrowException(Int32 hresult, ExcepInfo& excepInfo, UInt32 argErr, String message)
   at CallSite.Target(Closure, CallSite, ComObject)
   at System.Dynamic.UpdateDelegates.UpdateAndExecute1[T0,TRet](CallSite site, T0 arg0)
   at CallSite.Target(Closure, CallSite, Object)
   at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid1[T0](CallSite site, T0 arg0)
   at ConformU.TelescopeFacade.<AbortSlew>b__108_0() in J:\ConformU\ConformU\Conform\Devices\Facades\TelescopeFacade.cs:line 293
   at ConformU.DriverHostForm.ActionNoParameters(Action action) in J:\ConformU\ConformU\Conform\Devices\Facades\BaseClass\DriverHostForm.cs:line 163
   at ConformU.FacadeBaseClass.<>c__DisplayClass11_0.<MethodNoParameters>b__0() in J:\ConformU\ConformU\Conform\Devices\Facades\BaseClass\FacadeBaseClass.cs:line 230
   at System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Forms.Control.InvokeMarshaledCallbacks()
--- End of stack trace from previous location ---
   at System.Windows.Forms.Control.MarshaledInvoke(Control caller, Delegate method, Object[] args, Boolean synchronous)
   at System.Windows.Forms.Control.Invoke(Delegate method, Object[] args)
   at System.Windows.Forms.Control.Invoke(Action method)
   at ConformU.FacadeBaseClass.MethodNoParameters(Action action) in J:\ConformU\ConformU\Conform\Devices\Facades\BaseClass\FacadeBaseClass.cs:line 230
   at ConformU.TelescopeFacade.AbortSlew() in J:\ConformU\ConformU\Conform\Devices\Facades\TelescopeFacade.cs:line 293
   at ConformU.TelescopeTester.AbortSlew(String testName) in J:\ConformU\ConformU\Conform\Devices\TelescopeTester.cs:line 8259
   at ConformU.TelescopeTester.TelescopeOptionalMethodsTest(OptionalMethodType testType, String testName, Boolean canTest) in J:\ConformU\ConformU\Conform\Devices\TelescopeTester.cs:line 5687
15:06:33.946 AbortSlew                           ISSUE    Unexpected error: Unknown  Action is not implemented in this driver.

I **_expected to get here an ok_**

Peter-Simpson, 2024-05-01:
Hi Charles,

Apologies for the long delay in replying, I lost track of this issue. I think there are two issues here: 

1. Your AbortSlew method is throwing an ASCOM.ActionNotimplementedException instead of the expected ASCOM.MethodNotimplementedException.
2. There is a Conform bug that you haven't come across yet where Conform won't accept ASCOM.MethodNotImplementedException as a valid response to AbortSlew().

A fix for 2. will be in the next Conform Universal release.

Thanks for reporting the issue.

Best wishes, Peter

Peter-Simpson, 2024-09-26:
Closing because the fixed version has been released.

closed 2024-09-26 by Peter-Simpson