ASCOMInitiative/ConformU #15: Testing with the latitude set to the southern hemisphere causes errors

nigeldun, 2024-05-23:
Hi
Im testing out a new driver in ConformU and Ive hit a problem testing in the Southern hemisphere, Northern seems to work.

1 set my latitude to a negative value (-3) and ran the latest conformU code - 2.2.0-rc.3 (Build 26636.58f2f3d79821bd5fa5e11173c24f1d7e1397a2f8) and got ..

> 16:44:45.530 GetTestDeclination                  DEBUG    TRANSFORM: Site: Latitude: -03:10:50.3, Longitude: -94:11:18.2, Elevation: 403, Pressure: 1010, Temperature: 0, Refraction: False
16:44:45.531 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: -85:00:00.0, Azimuth: +176:27:59.9, Elevation: -00:21:31.4
16:44:45.531 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: -75:00:00.0, Azimuth: +169:21:52.7, Elevation: -07:25:11.1
16:44:45.532 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: -65:00:00.0, Azimuth: +162:01:56.4, Elevation: -14:21:52.3
16:44:45.533 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: -55:00:00.0, Azimuth: +154:14:15.7, Elevation: -21:04:11.3
16:44:45.533 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: -45:00:00.0, Azimuth: +145:43:43.0, Elevation: -27:23:12.8
16:44:45.534 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: -35:00:00.0, Azimuth: +136:14:19.3, Elevation: -33:07:40.8
16:44:45.534 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: -25:00:00.0, Azimuth: +125:31:26.6, Elevation: -38:03:18.7
16:44:45.535 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: -15:00:00.0, Azimuth: +113:27:10.8, Elevation: -41:52:55.6
16:44:45.535 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: -05:00:00.0, Azimuth: +100:08:58.7, Elevation: -44:18:22.9
16:44:45.536 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: +05:00:00.0, Azimuth: +86:06:24.7, Elevation: -45:05:09.3
16:44:45.536 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: +15:00:00.0, Azimuth: +72:06:22.7, Elevation: -44:07:57.1
16:44:45.536 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: +25:00:00.0, Azimuth: +58:54:42.4, Elevation: -41:33:12.1
16:44:45.537 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: +35:00:00.0, Azimuth: +46:58:40.8, Elevation: -37:36:03.8
16:44:45.537 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: +45:00:00.0, Azimuth: +36:23:46.7, Elevation: -32:34:44.5
16:44:45.538 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: +55:00:00.0, Azimuth: +27:01:03.8, Elevation: -26:46:10.7
16:44:45.538 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: +65:00:00.0, Azimuth: +18:35:35.0, Elevation: -20:24:19.6
16:44:45.539 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: +75:00:00.0, Azimuth: +10:51:23.0, Elevation: -13:40:10.9
16:44:45.542 GetTestDeclination                  DEBUG    TRANSFORM: RA: +10:34:59.25, Declination: +85:00:00.0, Azimuth: +03:33:27.6, Elevation: -06:42:28.1
16:44:45.542 GetTestDeclination                  DEBUG    Test RightAscension: +10:34:59.25, Lowest declination above horizon: +1.7976931348623157E+308:00:00.0, Test Declination: -8.988465674311579E+307:00:00.0 found in 12.0ms.
16:44:45.543 SlewScope                                    About to set Tracking property to true
16:44:45.613 SlewScope                           DEBUG    Slewing asynchronously to Hour angle -9.00 +10:34:59.25 -8.988465674311579E+307:00:00.0
16:44:45.613 SlewScope                                    About to call SlewToCoordinatesAsync method, RA: +10:34:59.25, Declination: -8.988465674311579E+307:00:00.0
16:44:45.615 CheckProperties                     ISSUE    Exception when testing device: SlewToCoordinatesAsync Declination set - '-8.98846567431158E+307' is an invalid value. The valid range is: -90 - +90.
16:44:45.617 CheckProperties                     DEBUG    System.Runtime.InteropServices.COMException (0x80040401): SlewToCoordinatesAsync Declination set - '-8.98846567431158E+307' is an invalid value. The valid range is: -90 - +90.
   at Microsoft.CSharp.RuntimeBinder.ComInterop.ComRuntimeHelpers.CheckThrowException(Int32 hresult, ExcepInfo& excepInfo, UInt32 argErr, String message)
   at CallSite.Target(Closure, CallSite, ComObject, Object, Object)
   at CallSite.Target(Closure, CallSite, Object, Object, Object)
   at ConformU.TelescopeFacade.<SlewToCoordinatesAsync>b__120_0(Object i, Object j) in J:\ConformU\ConformU\Conform\Devices\Facades\TelescopeFacade.cs:line 353
   at ConformU.DriverHostForm.Action2Parameters(Action`2 action, Object parameter1, Object parameter2) in J:\ConformU\ConformU\Conform\Devices\Facades\BaseClass\DriverHostForm.cs:line 197
   at ConformU.FacadeBaseClass.<>c__DisplayClass13_0.<Method2Parameters>b__0() in J:\ConformU\ConformU\Conform\Devices\Facades\BaseClass\FacadeBaseClass.cs:line 272
   at System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Forms.Control.InvokeMarshaledCallbacks()
--- End of stack trace from previous location ---
   at System.Windows.Forms.Control.MarshaledInvoke(Control caller, Delegate method, Object[] args, Boolean synchronous)
   at System.Windows.Forms.Control.Invoke(Delegate method, Object[] args)
   at System.Windows.Forms.Control.Invoke(Action method)
   at ConformU.FacadeBaseClass.Method2Parameters(Action`2 action, Object parameter1, Object parameter2) in J:\ConformU\ConformU\Conform\Devices\Facades\BaseClass\FacadeBaseClass.cs:line 272
   at ConformU.TelescopeFacade.SlewToCoordinatesAsync(Double rightAscension, Double declination) in J:\ConformU\ConformU\Conform\Devices\Facades\TelescopeFacade.cs:line 353
   at ConformU.TelescopeTester.SlewScope(Double pRa, Double pDec, String pMsg) in J:\ConformU\ConformU\Conform\Devices\TelescopeTester.cs:line 7296
   at ConformU.TelescopeTester.TestOffsetRate(String testName, Double testHa, Axis testAxis, Double expectedRate) in J:\ConformU\ConformU\Conform\Devices\TelescopeTester.cs:line 7927
   at ConformU.TelescopeTester.TestDeclinationOffsetRates(String testName, Double hourAngle) in J:\ConformU\ConformU\Conform\Devices\TelescopeTester.cs:line 7871
   at ConformU.TelescopeTester.CheckProperties() in J:\ConformU\ConformU\Conform\Devices\TelescopeTester.cs:line 937
   at ConformU.ConformanceTestManager.TestDevice(Int32 numberOfTestCycles) in J:\ConformU\ConformU\Conform\ConformanceTestManager.cs:line 252

The routine involved appears to be GetTestDeclinationHalfwayToHorizon() and looking at it it will never get a lowestDeclinationAboveHorizon thats not double.MaxValue in the south. Theres also a test of testDeclination being less than double.MaxValue which will always be true as its set to 0

Peter-Simpson, 2024-05-24:
Reopening pending testing, because GitHub automatically closed the issue.

nigeldun, 2024-07-12:
Southern hemisphere problems seem to have all been fixed in the ‘custom’ versions of conformu supplied to Wes. Thanks.

commit f5a7e767 (2024-05-23): Fix for bug in test declination algorithm. Fixes #15 - subject to further testing.
closed 2024-07-12 by nigeldun