ASCOMInitiative/ConformU #16: Value checks trigger even on error responses

RReverser, 2024-08-19:
Latest ConformU "Check Alpaca Protocol" is returning new errors that I believe I haven't seen in the past on my driver:

```
11:38:53.118 GET CCDTemperature            ISSUE  A JSON parameter of name Value was expected, but was not found in the JSON response. This test is case sensitive.
11:38:53.119 GET CCDTemperature                     Response: {"ClientTransactionID":67890,"ServerTransactionID":3553,"ErrorNumber":1024,"ErrorMessage":"Property or method not implemented"}
```

Is `Value` field really necessary even on error responses? I could set it to some dummy value, but it seems useless - if anything, I'd expect the tests to check that it's _never_ present on error responses?

RReverser, 2024-08-19:
> Here is a new installer, please can you let me know if it fixes the issue? [download.ascom-standards.org/beta/ConformU(3.0.0+30860)Setup.exe](https://download.ascom-standards.org/beta/ConformU(3.0.0+30860)Setup.exe)

Thanks, it does.

closed 2024-08-19 by RReverser

Peter-Simpson, 2024-08-19:
Great, thanks for confirming!

Best wishes, Peter