ASCOMInitiative/ConformU #17: Device passes validation when tracing and debugging are enabled, but not when they are disabled

rickbassham, 2024-08-24:
Attached is the log from running ConformU with debugging and tracing both enabled and disabled. 

With no changes to the device, we succeed when debugging and tracing are enabled:

```
10:06:01.503 Congratulations, no errors, warnings or issues found: your driver passes ASCOM validation!!          
```

but fail when they are disabled:

```
10:07:09.257 Your device had 6 issues, 0 errors and 0 configuration alerts          
10:07:09.257                                              
10:07:09.257 Issue Summary                                
10:07:09.257 CalibratorOn                        ISSUE    CalibratorOn with brightness 0 was unsuccessful - the returned CalibratorState was 'Ready' instead of 'Ready'. The synchronous operation took 0.0 seconds
10:07:09.257 CalibratorOn                        ISSUE    CalibratorOn with brightness 63 was unsuccessful - the returned CalibratorState was 'Ready' instead of 'Ready'. The synchronous operation took 0.0 seconds
10:07:09.257 CalibratorOn                        ISSUE    CalibratorOn with brightness 127 was unsuccessful - the returned CalibratorState was 'Ready' instead of 'Ready'. The synchronous operation took 0.0 seconds
10:07:09.257 CalibratorOn                        ISSUE    CalibratorOn with brightness 191 was unsuccessful - the returned CalibratorState was 'Ready' instead of 'Ready'. The synchronous operation took 0.0 seconds
10:07:09.258 CalibratorOn                        ISSUE    CalibratorOn with brightness 255 was unsuccessful - the returned CalibratorState was 'Ready' instead of 'Ready'. The synchronous operation took 0.0 seconds
10:07:09.258 CalibratorOff                       ISSUE    CalibratorOff was unsuccessful - the returned CalibratorState was 'Off' instead of 'Ready'. The synchronous operation took 0.0 seconds
```

[ascom.conformu.0928.557470.txt](https://github.com/user-attachments/files/16736858/ascom.conformu.0928.557470.txt)

Peter-Simpson, 2024-08-25:
Hi Rick,

Thanks for reporting the issue, which I can reproduce here. It occurs when calibrators change state in under 1 second and this wasn't covered in my test setup, which has a longer state change time.

[Updated Conform Universal](https://download.ascom-standards.org/beta/ConformU(3.0.1+31049)Setup.exe)

Please let me know if this solves the problem.

Best wishes, Peter

rickbassham, 2024-08-29:
Do you have a mac build for this version?

Peter-Simpson, 2024-08-29:
Hi Rick,

I've just published a new release, including a MacOS version, to GitHub [ConformU 3.1.0](https://github.com/ASCOMInitiative/ConformU/releases).

Best wishes, Peter

Peter-Simpson, 2024-08-29:
Closing because the fix is released.