ASCOMInitiative/ConformU #23: ConformU doesn't accept URLs with hostname

RReverser, 2024-12-13:
The CLI docs say:

> $ ~/conformu/conformu alpacaprotocol --help
> Description:
>   Check the specified Alpaca device for Alpaca protocol conformance
> 
> Usage:
>   conformu alpacaprotocol <Alpaca_URI> [options]
> 
> Arguments:
>   <Alpaca_URI>  The device's Alpaca URI (e.g. http://host:port/api/v1/[DeviceType]/[DeviceNumber]).
>                 The ASCOM device type will be inferred from the supplied URI.

However, counter-intuitively, trying to pass a URL with host fails:

```
$ ~/conformu/conformu alpacaprotocol http://localhost:32323/api/v1/telescope/0

Unable to identify the device identifier as either a COM ProgID or an Alpaca URI. Is there a typo in the device identifier?

The Alpaca URI argument is invalid:
```

It seems only passing IP address (127.0.0.1 in this case) is supported, but both should probably be accepted.

Peter-Simpson, 2024-12-14:
Hi Ingvar, I've fixed #23 and #24. Can you let me know which Linux flavour you are using so I can send you a build to test? The options are x64, arm32 and arm64, thanks, Peter

RReverser, 2024-12-15:
That's okay, I can just use `dotnet run` from the latest commit. I'll try tomorrow.

RReverser, 2024-12-16:
Thanks, latest commit works as expected now.

closed 2024-12-16 by RReverser

Peter-Simpson, 2024-12-17:
Great, thanks for letting me know.

RReverser, 2024-12-17:
Can you publish 4.0.1 please?

Peter-Simpson, 2024-12-17:
I’ve just published 4.1.0. 

 

Best wishes, Peter

RReverser, 2024-12-17:
Ah thanks. I think from semver point of view, a patch version here would be enough, since those are just bugfixes, but either one is fine :)

RReverser, 2024-12-17:
For now used in the simulators project: https://github.com/ASCOMInitiative/ASCOM.Alpaca.Simulators/pull/34

Looking good and useful: https://github.com/RReverser/ASCOM.Alpaca.Simulators/actions/runs/12379069796/job/34552397380#step:7:361

Peter-Simpson, 2024-12-18:
I'm glad it's working well in your CI checks, thanks for letting me know!

I'm surprised the OmniSim CameraState property reports "Download" state rather than "Idle" state at startup, which is why the AbortExposure test fails later in the results. I've never seen that in my use of the OmniSim to date, and it doesn't appear in the Windows version of 0.4.1-rc.1.

RReverser, 2024-12-18:
Yeah it seems there are couple of devices with issues in latest omnisim from git. Haven't seen them in the past either, but then, both Conform and Omnisim have changed meanwhile. https://github.com/ASCOMInitiative/ASCOM.Alpaca.Simulators/issues/33

RReverser, 2024-12-18:
Hmm, actually... maybe that particular issue is due to `conformu alpacaprotocol ...` leaving the device in "dirty" state? It seems AlpacaProtocolTestManager is testing the `ImageArray` property, which I guess could trigger a download instead of pure property access in the simulator? Just a guess though.

Peter-Simpson, 2024-12-19:
Can you try re-ordering the tests so the Conformance test runs before the Alpaca protocol test? If that works it would reinforce your suggestion about Alpaca protocol leaving the device in a dirty state.

RReverser, 2024-12-19:
Looks like that's it, it does pass the other way around: https://github.com/RReverser/ASCOM.Alpaca.Simulators/actions/runs/12413108501/job/34654290737

So now the question is only where the bug is...

RReverser, 2024-12-20:
I guess, as a separate question, why are those distinct test groups anyway? I'd expect a single command like `conformu http://...` would always run all tests (both Alpaca and Conformance tests together), which would also guarantee they don't conflict with each other.

I understand that conformance tests are a bit more universal, since they applie to COM devices as well, but if a device is detected to be Alpaca one (given a URL), I think there is no harm in just running everything?

RReverser, 2024-12-20:
> So now the question is only where the bug is...

Ah, it's right here:

https://github.com/ASCOMInitiative/ConformU/blob/5571b361c6014c673b4f3c7a9e9a57a60050a96c/ConformU/AlpacaProtocolTestManager.cs#L619-L622

this is only waiting for the exposing state to change, without accounting for the Download one. I'm going to send a PR.

Peter-Simpson, 2024-12-21:
I've committed some wait logic so the Alpaca protocol camera check will only exit with one of the two long-term camera states: Idle or Error, this should fix the ordering issue.

commit 5fb38236 (2024-12-14): Fix failure of the command line URI parser to accept host names, it only accepted IP addresses. Fixes #23.