ASCOMInitiative/ConformU #24: ConformU CLI doesn't work unless previously configured via GUI

RReverser, 2024-12-13:
I downloaded Linux version of ConformU CLI and tried running it against the simulator server. It fails with:

```
$ ~/conformu/conformu conformance http://127.0.0.1:32323/api/v1/telescope/0/

Cannot start test:
Alpaca interface version is not set
```

It seems it expects `conform.settings` to already exist and be pre-configured, instead of initializing settings with defaults and/or fetching Alpaca version and other necessary information from the provided URL.

This makes it hard to use ConformU in a fully automated CI environment, where you want to be able to just download it from Github releases and run against any server without access to the GUI.

RReverser, 2024-12-16:
Thanks, latest commit works as expected now.

closed 2024-12-16 by RReverser

Peter-Simpson, 2024-12-17:
Great, thanks for reporting the issue and letting me know it's fixed.