ASCOMInitiative/ConformU #25: ConformU fails on pre-async devices

RReverser, 2024-12-20:
Running Alpaca Protocol test against a device that doesn't implement new `connect`/`disconnect`/`connecting` endpoints eventually fails with:

```
18:22:50.905                                     ERROR    ASCOM.NotImplementedException: Property or method not implemented.
   at ASCOM.Alpaca.Clients.DynamicClientDriver.SendToRemoteDevice[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String uriBase, Boolean strictCasing, ILogger logger, String method, Dictionary`2 parameters, HttpMethod httpMethod, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 898
   at ASCOM.Alpaca.Clients.DynamicClientDriver.GetValue[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String URIBase, Boolean strictCasing, ILogger logger, String method, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 488
   at ASCOM.Alpaca.Clients.DynamicClientDriver.GetValue[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String URIBase, Boolean strictCasing, ILogger logger, String method, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 467
   at ASCOM.Alpaca.Clients.AlpacaDeviceBaseClass.get_Connecting() in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\AlpacaDevices\BaseClass\AlpacaDeviceBaseClass.cs:line 257
   at ConformU.AlpacaProtocolTestManager.<>c__DisplayClass53_0.<TestConnect>b__0() in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 503
   at ConformU.AlpacaProtocolTestManager.WaitWhile(String actionName, Func`1 waitFunction, Int32 pollInterval, Int32 timeoutSeconds, Func`1 statusString) in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 2952
   at ConformU.AlpacaProtocolTestManager.TestConnect(IAscomDeviceV2 device) in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 503
   at ConformU.AlpacaProtocolTestManager.TestCamera() in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 529
   at ConformU.AlpacaProtocolTestManager.TestAlpacaProtocol() in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 285
18:22:50.905  
18:22:50.906 Found 1 error, 0 issues and 11 information messages. 
18:22:50.906  
18:22:50.911 Error Summary 
18:22:50.913  ==> ASCOM.NotImplementedException: Property or method not implemented.
   at ASCOM.Alpaca.Clients.DynamicClientDriver.SendToRemoteDevice[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String uriBase, Boolean strictCasing, ILogger logger, String method, Dictionary`2 parameters, HttpMethod httpMethod, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 898
   at ASCOM.Alpaca.Clients.DynamicClientDriver.GetValue[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String URIBase, Boolean strictCasing, ILogger logger, String method, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 488
   at ASCOM.Alpaca.Clients.DynamicClientDriver.GetValue[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String URIBase, Boolean strictCasing, ILogger logger, String method, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 467
   at ASCOM.Alpaca.Clients.AlpacaDeviceBaseClass.get_Connecting() in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\AlpacaDevices\BaseClass\AlpacaDeviceBaseClass.cs:line 257
   at ConformU.AlpacaProtocolTestManager.<>c__DisplayClass53_0.<TestConnect>b__0() in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 503
   at ConformU.AlpacaProtocolTestManager.WaitWhile(String actionName, Func`1 waitFunction, Int32 pollInterval, Int32 timeoutSeconds, Func`1 statusString) in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 2952
   at ConformU.AlpacaProtocolTestManager.TestConnect(IAscomDeviceV2 device) in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 503
   at ConformU.AlpacaProtocolTestManager.TestCamera() in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 529
   at ConformU.AlpacaProtocolTestManager.TestAlpacaProtocol() in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 285
  Response: 
```

Same happens in conformance tests:

```
18:32:09.615 [Lib] Connecting 1 JSON response: {"ClientTransactionID":5,"ServerTransactionID":61,"ErrorNumber":1024,"ErrorMessage":"Property or method not implemented."}
18:32:09.615 [Lib] Connecting 1 Received an Alpaca error: AlpacaErrorCodeBase. Error message: Property or method not implemented.
18:32:09.615 [Lib] Connecting 1 Received Alpaca error code: AlpacaErrorCodeBase (0x0400), the equivalent COM error HResult error code is -2147220480 (0x80040400)
18:32:09.615 [Lib] Connecting 1 Alpaca member not implemented error, throwing NotImplementedException - ErrorMessage: "Property or method not implemented.", ErrorNumber: 0x80040400
18:32:09.616 [Lib] Connecting 1 Boolean Property or method not implemented.
18:32:09.616 [Lib] Connecting 1 Exception: ASCOM.NotImplementedException: Property or method not implemented.
   at ASCOM.Alpaca.Clients.DynamicClientDriver.SendToRemoteDevice[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String uriBase, Boolean strictCasing, ILogger logger, String method, Dictionary`2 parameters, HttpMethod httpMethod, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 898
18:32:09.616 ConnectToDevice                     ISSUE    Connection exception - further testing abandoned: Property or method not implemented.
18:32:09.618 ConnectToDevice                     DEBUG    ASCOM.NotImplementedException: Property or method not implemented.
   at ASCOM.Alpaca.Clients.DynamicClientDriver.SendToRemoteDevice[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String uriBase, Boolean strictCasing, ILogger logger, String method, Dictionary`2 parameters, HttpMethod httpMethod, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 898
   at ASCOM.Alpaca.Clients.DynamicClientDriver.GetValue[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String URIBase, Boolean strictCasing, ILogger logger, String method, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 488
   at ASCOM.Alpaca.Clients.DynamicClientDriver.GetValue[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String URIBase, Boolean strictCasing, ILogger logger, String method, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 467
   at ASCOM.Alpaca.Clients.AlpacaDeviceBaseClass.get_Connecting() in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\AlpacaDevices\BaseClass\AlpacaDeviceBaseClass.cs:line 257
   at ConformU.DeviceTesterBaseClass.Connect() in J:\ConformU\ConformU\Conform\DeviceTesterBaseClass\DeviceTesterBaseClass.cs:line 758
   at ConformU.ConformanceTestManager.TestDevice(Int32 numberOfTestCycles) in J:\ConformU\ConformU\ConformanceTestManager.cs:line 234
18:32:09.618 
18:32:09.618 ConnectToDevice                              Cannot connect to device, further tests abandoned.
```

I don't think this is intended - ConformU is supposed to work against pre-7.0 synchronous devices as well, right?

RReverser, 2024-12-20:
Speaking of, I have some thoughts about possibly simplifying Alpaca API in regards to async, after having gone through the effort of implementing an alternative Alpaca implementation. If you'd be ever up for a short discussion - like 20-30 min Meet call - would love to chat and share my experience.

Peter-Simpson, 2024-12-21:
The place where the code is failing is within an area that only executes when the interface version from the device indicates that the device supports Platform 7 interface members. Please can you make sure you are using the bug-fix ConformU version that I sent you because it has extended interface version checks. Please can you also check what interface version is being returned by your device and post the whole conformance log if there is still an issue.

RReverser, 2024-12-21:
Ah, thank you, that's likely it.

In this case I bumped the driver version but didn't implement some endpoints, as I assumed clients would still check the `/can...` endpoints like `/canasync` before invoking the new async methods.

I can see no `can` counterpart exists for connect/disconnect, which makes sense as you need to be connected for `can` checks to work... but I wonder if there should be a fallback anyway - if `connect` returns "not implemented", maybe client should switch to the old `PUT connected` instead?

Peter-Simpson, 2024-12-21:
The InterfaceVersion property is our mechanic for determining which methods are available, if the client checks this they will know whether or not new features like async connect are available. 

We believe it is up to driver / Alpaca device authors to implement in line with the interface specifications and I developed ConformU to help with this. Non-conformant devices just add to everyone's support burden and create a bad user experience. It should not be up to clients to adapt to incorrect device behaviour.

RReverser, 2024-12-21:
But in that case what are methods like `/switch/*/canasync`, `/telescope/*/canslewaltazasync` etc for? It seems they're supposed to allow for optional implementation of async methods - otherwise, if interface version was enough to determine whether device is capable of async operations, then those `can` indicators would be obsolete?

Peter-Simpson, 2024-12-21:
InterfaceVersion determines whether or not a mandatory or optional member exists at all in the interface. Mandatory methods must be implemented, when present in a given interface version, so InterfaceVersion is a reliable indicator of whether the member is functional and can be called by clients.

The async connection mechanic (Connect, Disconnect, Connecting) are mandatory in Platform 7 interfaces so must be present and must work when InterfaceVersion reports a Platform 7 interface. Consequently there is no need for a Can property.

The Can properties enable clients to determine when optional methods are implemented without calling them. This is so that implementation of methods that could have dangerous or unexpected or unwanted effects if called, can be detected without the risk of something untoward happening e.g. the scope slewing or a device controlled by a switch turning on.

RReverser, 2024-12-21:
Ah interesting, makes sense.

I think this misunderstanding could be qualified as a documentation issue - https://ascom-standards.org/api7/ does not really mark optional vs mandatory methods, so it's easy to get confused. I think I did have issues with that in the past, even for regular synchronous methods and had to dig into ASCOM docs to figure out which is which.

I know ASCOM's interface docs at least mark which methods might throw "not implemented" exception, so I assume those count as optional ones, but it would be useful to reflect that in the Alpaca docs somehow.

Maybe via method tags? Or just add `(optional)` to the end of the `title`. Anywhere where it could be prominent and, ideally, machine-readable, would be good.

Peter-Simpson, 2024-12-22:
I agree it would be more convenient if all information, including optional/mandatory status and expected exceptions, was duplicated in the Alpaca API definition, but we are concerned about the ongoing maintenance burden of keeping these aligned. It's already problematic keeping the OmniSim Swagger implementation and the Platform Help aligned to the [Master Interface Definition](https://ascom-standards.org/newdocs/) document. However, I'm keen to retain the information that is already present.

As of Platform 7 we are asking people to use the Alpaca API reference to get Alpaca specific HTTP syntax and variable types and to use the master interface document as the source of behavioural requirements.

RReverser, 2024-12-22:
Hm I see. I assumed the plan was to eventually migrate off COM interface as primary source of truth, keeping it as a backward compatibility interface for Alpaca, in which case eventually switching to Alpaca docs as primary ones would make more sense.

> Speaking of, I have some thoughts about possibly simplifying Alpaca API in regards to async, after having gone through the effort of implementing an alternative Alpaca implementation. If you'd be ever up for a short discussion - like 20-30 min Meet call - would love to chat and share my experience.

Just in case this was missed, would you be open to an informal chat? Happy to do it over a pint too if you're ever around London.

Peter-Simpson, 2024-12-22:
> I assumed the plan was to eventually migrate off COM interface as primary source of truth, keeping it as a backward compatibility interface for Alpaca, in which case eventually switching to Alpaca docs as primary ones would make more sense.

The driver behind Bob creating the platform neutral master docs was our desire to create a realisation independent definition of the ASCOM interfaces from which realisations in other technologies can be derived. However, some questions have come from people who treat the master document as platform specific e.g. COM developers who need to differentiate between single and double precision types but who find that all floating point types in the master docs are labelled "float". 

We have an agenda topic for our next meeting to discuss how we describe the platform neutral and platform specific documents

> Just in case this was missed, would you be open to an informal chat? Happy to do it over a pint too if you're ever around London.

Thanks for the offer, I live in Horsham West Sussex so London would be a stretch for me, perhaps a video call on Monday 22nd or Tuesday 23rd evening? If you are free let me know and I'll send a Zoom invitation.

I've closed the issue because its resolved but am happy keep discussion going here if you wish.

closed 2024-12-22 by Peter-Simpson

RReverser, 2024-12-22:
> Thanks for the offer, I live in Horsham West Sussex so London would be a stretch for me, perhaps a video call on Monday 22nd or Tuesday 23rd evening? 

Heh, I live at the edge of Surrey / West Sussex, but I don't drive, and public transport aside from the train line tobLondon is not great in these areas, so London usually ends up as the default meeting spot. Maybe someday in the future then :)

Yes, a video call on either evening sounds good. Can you please send an invitation with your preferred time to me@rreverser.com?

I'm not an official vendor or anything, but hoping it will be a useful discussion for both of us.

Peter-Simpson, 2024-12-23:
I've sent an invitation for 20:000 today. Can you let me know what it is that you are "not an official vendor" of so I can have a look before our meeting?

RReverser, 2024-12-23:
> Can you let me know what it is that you are "not an official vendor" of

Oh, I just meant not a vendor in general :)