ASCOMInitiative/ConformU #26: ConformU incorrectly flags issue when windspeed is 0.0 and winddirection is unsupported

arobicha, 2025-01-20:
While running a conformance test with our Observing Conditions capable Alpaca device using ConformU v4.1.0 (Build 34702.5571b36) we get the following issue:

```txt
11:59:24.331 WindSpeed                           ISSUE    When wind speed is reported as 0.0, wind direction should also be reported as 0.0, it is actually reported as NaN
```

Our sensor reports Wind Direction is not implemented, which results in this line producing a `windDirection` of `BAD_VALUE` (i.e. `NaN`)

See: [ObservingConditionsTester.cs#L326](https://github.com/ASCOMInitiative/ConformU/blob/38492a2771d4d99c125dd748b0e67f30d2ab23bd/ConformU/Conform/ObservingConditionsTester.cs#L326C13-L326C138)

```csharp
windDirection = TestDouble(PROPERTY_WINDDIRECTION, ObservingConditionsProperty.WindDirection, 0.0, 360.0, Required.Optional);
```

See: [ObservingConditionsTester.cs#L330](https://github.com/ASCOMInitiative/ConformU/blob/38492a2771d4d99c125dd748b0e67f30d2ab23bd/ConformU/Conform/ObservingConditionsTester.cs#L330C13-L337C14)
```csharp
// Additional test to confirm that the reported direction is 0.0 if the wind speed is reported as 0.0
if ((windSpeed == 0.0))
{
    if ((windDirection == 0.0))
        LogOk(PROPERTY_WINDSPEED, "Wind direction is reported as 0.0 when wind speed is 0.0");
    else
        LogIssue(PROPERTY_WINDSPEED, string.Format("When wind speed is reported as 0.0, wind direction should also be reported as 0.0, it is actually reported as {0}", windDirection));
}
```

arobicha, 2025-01-20:
Note the response to WindDirection from the sensor:
```json
{
	"Value": 0,
	"ClientTransactionID": 0,
	"ServerTransactionID": 58,
	"ErrorNumber": 1024,
	"ErrorMessage": "Not implemented."
}
```

And the response from WindSpeed:
```json
{
	"Value": 0,
	"ClientTransactionID": 0,
	"ServerTransactionID": 60,
	"ErrorNumber": 0,
	"ErrorMessage": ""
}
```

arobicha, 2025-01-21:
Hi Peter,

Thanks for the ridiculously fast turn around. I can confirm Version 4.2.0-rc.2 corrects the issue.

Cheers,
-Adam

Peter-Simpson, 2025-01-23:
Great, thanks for letting me know Adam.

Best wishes, Peter

commit 5648b8c1 (2025-01-21): ObservingConditions - Don't check for wind direction being reported as 0.0 when winds peed is 0.0 if wind direction is not implemented. Fixes #26.
closed 2025-01-23 by Peter-Simpson