ASCOMInitiative/ConformU #27: Guide pulse testing

ClutchplateDude, 2025-02-02:
I work on the OpenAstroTech mounts and our Conformance test has some issues that I don't think are issues...

Why do the Guide Pulse tests expect the RA/DEC coordinates to change?

My understanding is that the whole point of guiding is to keep the RA/DEC coordinate constant. Why do these tests expect coordinate changes?

Peter-Simpson, 2025-02-03:
Guiding compensates for:

1. "Apparent" movement of the target due to atmospheric turbulence changing the refractive index of the air column along the line of site to the target.
2. Mechanical issues in the mount drive train e.g. periodic error.

In both cases the target's coordinates on the sky change because either refraction moves the image to different coordinates or the mount is no longer pointing at the coordinates that it is reporting. 

When the guiding system detects a movement of the target it is impossible for it to know whether the target moved because of atmospheric turbulence or because of mechanical issues, and the astronomical community convention is to treat the movement as real coordinate change from turbulence because there is nothing that the average observer can do about that. 

In many mounts mechanical issues can be addressed, for example through periodic error correction in the mount control software.

For this reason Conform expects mount coordinates to change when guiding.

ClutchplateDude, 2025-03-06:
So let's say that the mount is pointing at (RA1, DEC1) and tracking speed is slightly off (or some other non-periodic problem) and guiding corrects this and keeps the mount pointed at (RA1, DEC1) so that imaging remains framed correctly. After a while the mount is now reporting that it points at (RA2, DEC2) since all the guiding pulse have caused that. Now the GOTO of the mount is completely off by the difference (RA2-RA1, DEC2-DEC1), even though the mount is STILL pointing at RA1,DEC1. Is that the expected behavior?

Peter-Simpson, 2025-03-07:
Guiding is primarily intended to compensate for random or periodic errors, such as seeing or periodic error, which cause the target to move within a limited space around the nominal target coordinates. In these scenarios the mount's reported coordinates change slightly as guiding kicks in but remain a good approximation for the target's sky coordinates, assuming the mount has a good pointing model in the first place.

In your example of a poorly tracking mount with non random / non periodic errors, yes the mount's, reported coordinates can start to differ from the target's sky coordinates but this is due to mount defects or a poor pointing model rather than being an effect of guiding.

ClutchplateDude, 2025-03-07:
Ok, thank you.

commit e2539973 (2024-12-14): Ensure that the Alpaca API version number is correctly initialised from the supplied Alpaca URI. Fixes #27.
closed 2025-03-07 by ClutchplateDude