ASCOMInitiative/ConformU #28: Exception when checking ALPACA protocol

marcocipriani01, 2025-04-27:
I just added support for ASCOM 7 features (Connect, Disconnect, DeviceState) to my custom dome ALPACA driver. They work as expected via `curl`, and also my driver used to pass conformance and protocol checks successfully before the update. Now, however, ConformU throws the following error when checking the ALPACA protocol for my driver. I suspect it's just a ConformU bug, since it's a stacktrace and not an issue.
I can make the error go away (and protocol checks finish successfully) if I set `Dome.InterfaceVersion` back to `2` in my driver. If I set it to `3`, I always get the stacktrace in ConformU.

[ConformU log.txt](https://github.com/user-attachments/files/19931322/ConformU.log.txt)

```
19:20:58.438 Error Summary 
19:20:58.440  ==> ASCOM.DriverException: Error calling method: Connecting, HTTP Completion Status: NotFound, Error Message:

   at ASCOM.Alpaca.Clients.DynamicClientDriver.SendToRemoteDevice[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String uriBase, Boolean strictCasing, ILogger logger, String method, Dictionary`2 parameters, HttpMethod httpMethod, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 1551
   at ASCOM.Alpaca.Clients.DynamicClientDriver.GetValue[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String URIBase, Boolean strictCasing, ILogger logger, String method, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 488
   at ASCOM.Alpaca.Clients.DynamicClientDriver.GetValue[T](UInt32 clientNumber, HttpClient client, Int32 timeout, String URIBase, Boolean strictCasing, ILogger logger, String method, MemberTypes memberType) in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\DynamicClientDriver.cs:line 467
   at ASCOM.Alpaca.Clients.AlpacaDeviceBaseClass.get_Connecting() in J:\ASCOMLibrary\ASCOM.Alpaca\ASCOM.Alpaca.Clients\AlpacaDevices\BaseClass\AlpacaDeviceBaseClass.cs:line 257
   at ConformU.AlpacaProtocolTestManager.<>c__DisplayClass53_0.<TestConnect>b__0() in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 503
   at ConformU.AlpacaProtocolTestManager.WaitWhile(String actionName, Func`1 waitFunction, Int32 pollInterval, Int32 timeoutSeconds, Func`1 statusString) in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 2952
   at ConformU.AlpacaProtocolTestManager.TestConnect(IAscomDeviceV2 device) in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 503
   at ConformU.AlpacaProtocolTestManager.TestDome() in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 734
   at ConformU.AlpacaProtocolTestManager.TestAlpacaProtocol() in J:\ConformU\ConformU\AlpacaProtocolTestManager.cs:line 293
  Response: 
```

Using ConformU version 4.1.0 (Build 34702.5571b36) on Windows 11 24H2.
The custom dome driver runs on an STM32 microcontroller connected to the network.

Let me know if you need more information.
Regards,
Marco Cipriani

Peter-Simpson, 2025-04-28:
Hi Marco,

I am unable to reproduce this issue using the Platform OmniSims. The issue appears related to the Connecting property in your driver, please can you check whether this looks like it is implemented correctly?

If the issue persists, please will you:
- Enable Debug logging, Display method calls and Trace Alpaca calls as shown in the first image below
- Run a Conformance check and attach the full log to a post.
- Set the Check Alpaca protocol configuration as shown in the second image below
- Re-run Check Alpaca protocol and attach the full log to a post.

Many thanks, Peter

[image]

[image]

marcocipriani01, 2025-05-04:
Hi Peter,
I repeated the conformance today and ConformU told me that I was missing the "connecting" property. For some reason, last week it kept throwing that exception instead of noticing the missing property, which I added and now everything works. Thanks anyway!
Marco